bambulab/BambuStudio #5318: Sort filaments by name

brianharwell, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Filaments are not sorted by name, and they are sorted inconsistently between screens.

In this screenshot, on the Prepare tab I expect:
- PETG to be sorted before PLA
- eSun to be sorted before Sunlu. It will sort before if I name it ESun but that is not the correct name
[image]

In this screenshot, the Bambu Lab filaments are at the top of the list and again I expect PETG to be sorted before PLA
[image]

If you click the button next to the filament to get the Filament Settings, the Bambu Lab PLA filaments are sorted with the other PLA filaments, but again PETG should be before PLA and eSun should be before Sunlu.
[image]

### Actual results

Filaments are sorted in an inconsistent and incoherent manner

### Expected results

Filaments should be sorted by Name in Ascending order in both screens (Prepare and Filament Settings)

1. PETG should be sorted before PLA
2. eSun should be sorted before Sunlu
3. Bambu Lab should be sorted with other filaments

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

brianharwell, 2024-12-04:
@DanBao-Bambu any update on this?

bambu123, 2024-12-16:
It has indeed been adjusted to sort by frequency. However, custom presets should retain alphabetical sorting.

Alan-Somebody, 2025-01-05:
I have over 50 different filaments all with their own custom pre-sets.  Having them listed in very odd orders causes me a lot of grief and time when I try to find them.  Below you will see what I mean (the 'P1S' ones have been split up).  Since mine are all custom pre-sets, @bambu123 , shouldn't they be listed alphabetically as that is what you said above should happen.

[image]

Alan-Somebody, 2025-01-05:
I have over 50 different filaments all with their own custom pre-sets.  Having them listed in very odd orders causes me a lot of grief and time when I try to find them.  Below you will see what I mean (the 'P1S' ones have been split up).  Since mine are all custom pre-sets, @bambu123 , shouldn't they be listed alphabetically as that is what you said above should happen.

[image]

When I go into the Filament editing screen, everything sorts alphabetically as you would expect it to, and it is perfect.  The filament pre-set list in the 'Prepare' work area needs sort alphabetically.  Can this be looked at?

[image]

brianharwell, 2025-01-23:
@bambu123 Can we get a status update on this?

lorenzofanchi, 2025-02-28:
This is very annoying indeed. Alphabetical order gives one a clear expectation of where to find the profile one's looking for.

brianharwell, 2025-03-21:
@bambu123 @DanBao-Bambu It’s been over three months without any update from Bambu regarding this issue. I submitted a support ticket and was told there’s nothing they can do and to follow up here instead. I’ve been patient, but waiting over three months to fix a dropdown is unreasonable, especially when there’s already a PR ready to be merged (https://github.com/bambulab/BambuStudio/pull/4700).

Can someone please explain the delay?