bambulab/BambuStudio #5319: stop print opens web site

UBaer21, 2024-11-17:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S

### How to reproduce

[image]
[image]
YES

### Actual results

opens Browser 
https://wiki.bambulab.com/en/software/bambu-studio/faq/live-view

### Expected results

Not open Browser website

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-18:
It is a mistake when your pop-up button is laid on the top of liveview error string, move the pop-up to left or right a little can resolve this question, I believe they will fix that next version.

UBaer21, 2024-11-18:
@JohnSmithCYM you are right that this is a click through, although the link below is not visible as a link 
[image]

They should not allow a click through to the underlaying element.