bambulab/BambuStudio #5320: Document Process for Adding New System Profiles

mkstephenson, 2024-11-17:
There are filament manufacturers out there who create filament profiles for their filaments on the Bambu printers and host them in various places (colorFabb upload them as 3MF files on printables.com (https://www.printables.com/model/438171-bambu-lab-profiles-x1c-x1e-and-p1s/files), extrudr as JSON files on their website (https://extrudr.com/shop-eu/page/material-profiles/?locale=en) and dasfilament on their blog as bbsflmt files (https://www.dasfilament.de/blog/update-filamentprofile-fuer-bambu-studio-pla/petg/tpu)).

However, when I look at the system filament profiles for my P1S in Bambu Studio, I see other filament manufacturers (Polymaker and eSUN) alongside the Bambu Lab filaments. This got me wondering, is there a process for companies to add their filaments to this list? If so, would it be possible to document this somewhere so that more filament manufacturers can add their own profiles to this list of system profiles in Bambu Studio? This would make using these filaments much easier for us users.

teambvd, 2024-12-10:
@mkstephenson you can head to File -> Import -> Configs, then select the downloaded presets, that should cover it 👍

mkstephenson, 2024-12-15:
@teambvd sadly this only works if the filament publisher has actually created filament profiles for Bambu Studio. I meant publishing the policy regarding system profiles themselves (the ones delivered with Bambu Studio) and how filament manufacturers can submit their presets for this list.

teambvd, 2024-12-17:
It's the same as for OrcaSlicer, insofar as I'm aware at least - nothing specific to Bambu Studio, but generically the same for Prusa / Orca as well, unless I've misunderstood somehow?