bambulab/BambuStudio #5321: Chamber temperature not displayed until after you send print

HomerSimpy, 2024-11-17:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

macOS

### Additional system information

MacBook Pro M3 Max, 48 GB
[image]

### Printer

Bambu X1C

### How to reproduce

Open Bambu Studio, got to device page, no chamber temp readout.

### Actual results

Same as above

### Expected results

Chamber Temp should be available at all times as it was previously, all temps should be. When printing ASA, ABS etc. pre heating of the chamber to 40 degrees BEFORE sending print is ideal. If you can't see the chamber temp you have no idea when it reaches 40. 

### Project file & Debug log uploads

[Temp Issue.3mf.zip](https://github.com/user-attachments/files/17792736/Temp.Issue.3mf.zip)
[Archive.zip](https://github.com/user-attachments/files/17792766/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-18:
[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50) has solved this bug, try it!

walterwongbbl, 2024-11-29:
Please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50

closed 2024-11-29 by walterwongbbl