bambulab/BambuStudio #5323: Flow Dynamics Calibration Tab Acting as Though the Printer is Disconnected

IsExtant, 2024-11-18:
### Bambu Studio Version

Bambu_Studio_win_public-v01.10.01.50-20241115162711

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11, version 23H2

### Printer

Bambu X1C

### How to reproduce

Open Bambu Studio.
Open Calibration Tab, Printer will briefly appear connected in the top left of main window. Then switch to "no printer" and calibration buttons will grey out.
Only started in the newest version.

### Actual results

[image]

### Expected results

You have greyed out calibration buttons and are blocked from preforming the calibration. Others have encountered this issue.

https://www.reddit.com/r/BambuLab/comments/1gpqypp/why_is_the_calibration_button_greyed_out_and_cant/

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-26:
@IsExtant Confirm it is a problem with A1 auto-calibration (due to it is a new feature worked with A1 firmware), it will be fixed in next version.

zhimin-zeng-bambulab, 2024-11-26:
we will fix it in the next version.

zhimin-zeng-bambulab, 2025-02-05:
It has been fixed in v1.10.2 Public Beta

closed 2025-02-05 by zhimin-zeng-bambulab