bambulab/BambuStudio #5326: AMS display doesn't show loaded filament correctly

ColinRedGriff, 2024-11-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OSX Sonoma Version 14.6.1

### Printer

P1S

### How to reproduce

Send a print to the printer

### Actual results

The filament loads from the AMS but Bambu Studios does not show the loaded filament path as it used to. See screen shot attached the filament from A1 in the AMD has successfully loaded but there is only a small section on the AMS diagram showing the green filament.
[image]

### Expected results

The line fron AMS A1 to the printhead in the diagram should be fully green

### Project file & Debug log uploads

[4x4 Multiboard Core Tile.3mf.zip](https://github.com/user-attachments/files/17800322/4x4.Multiboard.Core.Tile.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ColinRedGriff, 2024-11-18:
Additionally, it does not matter which filament is loaded the same display issue is present. If I stop and restart Bambu Studio while the job is printing the display is corrected. New screenshot with A2 filament loaded after Bambu Studios restart
[image]

walterwongbbl, 2024-11-27:
@ColinRedGriff 
If you drag or zoom the main window, will the page refresh again