bambulab/BambuStudio #5327: Camera not connecting

TSko1106, 2024-11-18:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Slice file
2. Send to printer
3. hit play button to view print via the camera from my computer
4. camera freezes at initializing and will not open

### Actual results

After the most recent update, i cannot get my camera to open from my computer on any of my printers.  It is stuck on initializing.

### Expected results

When I hit the play button it should connect to my camera so i can view my prints.

### Project file & Debug log uploads

Its all prints not just one in particular.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

markvrepo, 2024-11-19:
Same.  Worked earlier.  Hardware is X1C w/AMS.
Started print from BS.  Video was working.  Print finished.  Hit reprint on the printer itself.  Neither phone (android) or windows app will connect to the camera.

ttt888zzz, 2024-11-19:
HI[TSko1106](https://github.com/TSko1106)、[markvrepo](https://github.com/markvrepo)、[donhohler](https://github.com/donhohler)
Could you please go to the specified directory and upload the log file  to help check the issue? thank you.
Export the Bambu Studio log：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log
[image]

RaphTronic, 2024-11-22:
Same problem with my P1S.
Just upgraded from 1.9.1.66 to 1.10.1.50.
Started a previously successful print right after the update, and get no video, only the Bambu Lab logo and stuck at "Initializing..."
Screenshot attached.

[image]

0h-crap, 2024-11-22:
I have the same issue. I upgraded to 01.10.00.89 and it stuck on initializing.
I went back to version 01.08.04.51 and it worked.
I then went to version 01.09.07.52 and that worked as well.
When I went to version 01.10.00.89 it stopped working. Camera just sticks on Initializing.

Running on a Windows 11 PC

ollehu, 2025-01-03:
> I have the same issue. I upgraded to 01.10.00.89 and it stuck on initializing. I went back to version 01.08.04.51 and it worked. I then went to version 01.09.07.52 and that worked as well. When I [...]

Exactly the same for me. Rolled back to pre-01.10 for this reason.

RaphTronic, 2025-01-04:
This issue was fixed on my P1S by simply updating its firmware, per BL's recommendation: https://github.com/bambulab/BambuStudio/issues/5367