bambulab/BambuStudio #5330: Interface initial layer changed from 1.9.7 to 1.10 of Bambu Studio

Talmadge3000, 2024-11-19:
### Bambu Studio Version

1.10.1

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS 15.0.1

### Additional system information

Apple M2 Max
96 GB Ram

### Printer

Bambu P1S

### How to reproduce

1. Upgrade Bambu Studio to 1.10.1
2. Initial support layer is dense and monolithic(I think that is what it is called)
3. Downgrade to 1.9.7
4. Initial support layer is Concentric 

### Actual results

With Bambu Studio to 1.10.1 the Initial support layer is dense and monolithic(I think that is what it is called)
This is very hard to remove from the stackable bin I am trying to make.
With version 1.9.7 this interface is easy to remove. 

### Expected results

With version 1.9.7 this interface is easy to remove. 
I cannot figure out what to change in 1.10.1 to fix the issue or if this is a bug. 
[image]

### Project file & Debug log uploads

[StackableBin.3mf.zip](https://github.com/user-attachments/files/17807702/StackableBin.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-20:
[image]
[image]
I printed this amazing box with my A1, I only have PLA and PETG basic, and as the picture, it is good, and it seems the intial layer of interface is concentric, I noticed that you used PETG translucent? I will test before I get one...

Talmadge3000, 2024-11-20:
Recalibrated the printer and now it seems to be peeling easier. Not sure what happened. But I am happy.

closed 2024-11-26 by JohnSmithCYM