bambulab/BambuStudio #5331: Allow specifying object sequence when printing by layers

luder888, 2024-11-19:
**Is your feature request related to a problem? Please describe.**
When printing in "by layer" mode, it'd be nice to be able to specify the print sequence for the 1st layer. When you have a plate full of objects, certain objects might end up being external facing, and a perfect 1st layer is critical. While other objects might be hidden and the 1st layer isn't as critical. By having the ability to specifying the sequence when printing by layers, one can quickly observe a failed 1st layer, abandon the print quickly, without having to wait for all the object's 1st layer to be completed before abandoning the print, saving time and filament.

**Describe the solution you'd like**
The ability to specify print sequence when printing in "by layer" mode, just like how you can in the "by object" mode today.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

FallasAutomation, 2024-11-19:
Go to Process/object tab, click and drag the objects/assemblies in the order you want them to print.

[image]

[image]

luder888, 2024-11-19:
@FallasAutomation This only works when "print by object" is active. That order you showed is being ignored when int "print by layer" mode. Also I'm printing with a single filament.

FallasAutomation, 2024-11-19:
> @FallasAutomation This only works when "print by object" is active. That order you showed is being ignored when in "print by layer" mode. Also I'm printing with a single filament.

Oops! Somehow how I missed the glaring "by layer" mentioned everywhere. Ha!