bambulab/BambuStudio #5333: Brim must take precedence over supports

nnobuddy, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

1. Prepare, Support Painting => paint supports.
[image]

2. Prepare/Preview, Support => set parameters.
[image]

3. Prepare/Preview, Bed adhesion => set parameters.
[image]

### Actual results

See the gaps between the object and the support on the first layer. The gap between the object and the support is much bigger then the gap between the object and the brim.
[image]

Print fails with Plate Type = Smooth PEI Plate / Hi Temp Plate (noname), even if the plate is cleaned with isopropyl alcohol. 
[image]
[image]

### Expected results

Print should not fail.

Print does not fail with Textured PEI Plate. Unfortunately, the bottom surface of the object is not smooth.

### Project file & Debug log uploads

[Prsten04-17-18-19_2.zip](https://github.com/user-attachments/files/17807250/Prsten04-17-18-19_2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

JohnSmithCYM, 2024-11-19:
It seems that you should use 'Support first layer gap' to get what you want.
[image]

nnobuddy, 2024-11-19:
Thank you very much for your help.

Print fails either way (on higher Z value). Unfortunately, I can't use glue to increase adhesion. I'm going to investigate the thing... :-)

closed 2024-11-19 by nnobuddy

JohnSmithCYM, 2024-11-20:
Do you know bambu has just released a new plate, I bought one and use it with studio, it sticks a lot.(I have to say even it sticks too much because I can't remove some parts from it easily lol)

JohnSmithCYM, 2024-11-21:
The ShiningStar plate is a 3rd-party plate but it looks good :heart: ,maybe it could produce a good bottom surface.Good luck with your print.