bambulab/BambuStudio #5334: Supports became Buggy

mLihs, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.0.1

### Additional system information

Mac Book Pro - Apple M3 Pro

### Printer

P1S

### How to reproduce

Select the model  and Print
[Housing.3mf.zip](https://github.com/user-attachments/files/17810937/Housing.3mf.zip)

Used with Bambulab Basic PLA

### Actual results

the Problems that occurred after _1_9_7_52:

-  Hybrid Support
      - The Support, stick at 20mm layer height with 0.16mm z Distance on the object and is not Removable anymore.
      -  has some missing layers in the z Direction of the "normal support"
- The Regular Support  
    - Less Easy Removable 20mm layer height with 0.16mm z Distance
    
- Filament Color influence
   - it Seams that white filament have a different behavior dan black filament on overhangs with support. The White Filament starts wobbling where the Black filament look pretty good. 

    
  

### Expected results

Removable Supports

### Project file & Debug log uploads

[Housing.3mf.zip](https://github.com/user-attachments/files/17812499/Housing.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

JohnSmithCYM, 2024-11-20:
I print your 3mf and here is the support part:
[image]
The hybrid support top-z is not usable seems a bug, maybe they will fix it in a short time.

tangketan, 2024-11-25:
Thanks for reporting. We are looking into it.