bambulab/BambuStudio #5335: AMS color diagram not updated when switching filaments printing (BS)

UBaer21, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Printer

P1S AMS×3

### How to reproduce

Print with color change (object)

### Actual results

switched from blue AMS A to grey on AMS B
but still showing blue in the diagram
[image]
after clicking the second AMS
[image]

### Expected results

at least showing the color on the extruder. 
maybe also switching to the AMS that supply the color

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

RB-gatech, 2024-11-20:
Similar issue, but noticed the color does change where it goes into the printhead.  At the time of the image capture, A3(black was printing. But is shows A4(green) except where it goes into the printhead.

[image]

JohnSmithCYM, 2024-11-26:
The same with #5313 , #5292 ,it seems a layout problem, re-enter the Device tab or change size of form may help you get it before Studio fix it.