bambulab/BambuStudio #5336: Bambu printer missing from bambu studio

Zeds34, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 23H2

### Additional system information

AMD Ryzen 9 7950X 16-Core Processor 4.50 GHz, 64 g ram, video card RX7700XT 12 Go.

### Printer

X1C 2x AMS

### How to reproduce

This morning when opening Bambu Studio a message says to update the models and my X1C has disappeared from bambu studio, I am looking to add a printer there is no printer in the bambu lab brand selection.

[image]

### Actual results

The Bambu Lab brand has disappeared from the printer selection.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Expected results

Bambu brand printers should be displayed in printer additions. Bambu Lab should be there.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Lex74M, 2024-11-20:
I have the same issue and cannot set up a printer as my bambu lab p1s is no longer in the list of printer options.....
Help

tommyboy180, 2024-11-21:
I'm unable to replicate. 

https://github.com/user-attachments/assets/d1f62b3c-eab4-42fb-a644-611edfe499ca

ttt888zzz, 2024-11-22:
HI [Zeds34](https://github.com/Zeds34)、 [Lex74M](https://github.com/Lex74M)：Please try uninstalling and reinstalling BambuStudio
 Download the latest installation package, please visit：https://bambulab.cn/zh-cn/download/studio

Zeds34, 2024-11-22:
I reset and everything came back. Thanks for the support.

closed 2024-11-26 by JohnSmithCYM