bambulab/BambuStudio #5338: Font color barely visible on Linux dark theme

mzanetti, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.10, AppImage

### Additional system information

In some parts of the UI, the text isn't readable due to a dark on dark front.

I'm using Ubuntu with the dark style. Generally BambuStudio adapts well to, with the exception of some views:

The updates view (e.g. network plugin installed, Filament profile updates...):
[image]

The Objects view (note how only the currently selected one is readable):
[image]

### Printer

X1C

### How to reproduce

Change Ubuntu to use the dark style systemwide.
Start Bamub Studio from the AppImage.

### Actual results

Font is barely readable in some places.

### Expected results

Font is readable :)

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

mzanetti, 2024-11-21:
Found another one
[image]

MackBambu, 2024-11-24:
confirm

dng-sitrep, 2024-12-20:
same issue Ubuntu 24.10 x86_64 Yaru-dark [GTK2/3]
[image]

closed 2025-03-31 by MackBambu