bambulab/BambuStudio #5339: Finish Time useless when printing into next day. "+1" is not helpful.

FallasAutomation, 2024-11-19:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

Start a print that doesn't finish until the following day or multiple days later.

### Actual results

The Finish Time will show a "+1","+2", etc. and no other information. This "feature" is basically useless for long prints or even short prints that are started late in the evening.

[image]

### Expected results

The Finish Time needs more helpful information. Simply adding the date and time it will be done would solve it. For example, a seven hour print started on Nov 19th at 6:00PM (18:00) would show "Nov 20th 01:00" as the finish time. This tells you exactly the information you need.

### Project file & Debug log uploads

[Dry+Pods+Wide+&+Narrow,+Wide+with+Round+Hygrometer,+Desiccant+Trays,+and+Funnel+V4.1.zip](https://github.com/user-attachments/files/17821148/Dry%2BPods%2BWide%2B.%2BNarrow.%2BWide%2Bwith%2BRound%2BHygrometer.%2BDesiccant%2BTrays.%2Band%2BFunnel%2BV4.1.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DanBao-Bambu, 2024-11-20:
The reason why the end time section is not displayed may be due to the screen zoom ratio. If you can try changing the screen zoom ratio to solve this problem, we will continue to follow up on it.

vinaykapadia, 2025-02-10:
It doesn't even work right. It's currently 3:34pm, my print is only 2 hours long and is already halfway done, and it shows the +1 there.

[image]

FoxyWoxy12000, 2025-03-27:
How do I make it so the text for this actually shows up?

FallasAutomation, 2025-03-27:
> How do I make it so the text for this actually shows up?

Change the scale (DPI) in your Windows Settings to 125% and see if that will fix it.

vinaykapadia, 2025-03-27:
> Change the scale (DPI) in your Windows Settings to 125% and see if that will fix it.

Which is really silly. This should be fixed.

vinaykapadia, 2025-03-27:
Also I figured out why the +1 showed even though my print wasn't going into the next day. It's based on UTC. You can't set a timezone.