bambulab/BambuStudio #5340: Typography glitch in preview page

dadams-800r, 2024-11-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.1.1

### Additional system information

Model Name:	MacBook Pro
  Model Identifier:	MacBookPro18,1
  Model Number:	MK1E3LL/A
  Chip:	Apple M1 Pro
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	16 GB
  System Firmware Version:	11881.41.5
  OS Loader Version:	11881.41.5

Apple M1 Pro:

  Chipset Model:	Apple M1 Pro
  Type:	GPU
  Bus:	Built-In
  Total Number of Cores:	16
  Vendor:	Apple (0x106b)
  Metal Support:	Metal 3
  Displays:
DELL U2515H:
  Resolution:	2560 x 1440 (QHD/WQHD - Wide Quad High Definition)
  UI Looks like:	2560 x 1440 @ 60.00Hz
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
DELL U2518D:
  Resolution:	2560 x 1440 (QHD/WQHD - Wide Quad High Definition)
  UI Looks like:	2560 x 1440 @ 60.00Hz
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported

### Printer

Bambu Lab A1 W/AMS Lite

### How to reproduce

Go to device tab while printing, look at the Printin Progress section at the bottom of the screen. Note the line running through the progress text for layer number and remaining minutes

[image]

### Actual results

You will see the anomaly as described above

### Expected results

The text should be clear of anomalies 

### Project file & Debug log uploads

[iPhone Standby Mode Dock (designed with OVERWERK).3mf.zip](https://github.com/user-attachments/files/17823593/iPhone.Standby.Mode.Dock.designed.with.OVERWERK.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

reedy, 2024-11-25:
[image]

dadams-800r, 2024-11-25:
Thanks @reedy - good to know I'm not losing my mind :)

reedy, 2024-11-25:
> Thanks @reedy - good to know I'm not losing my mind :)

I was worried I had a monitor fault or something too... Resizing the window, no it's still there... xD

Not sure if this is a macOS specific bug, but I'm still on 14.7.1

JohnSmithCYM, 2024-11-26:
@reedy It's a bug and it is confirmed in some issues like #5299

reedy, 2024-11-26:
So this one is actually a dupe...

JohnSmithCYM, 2024-11-26:
@reedy It's really an annoying but not fatal bug, don't throw your screen away and wait for their fix haha.