bambulab/BambuStudio #5343: 底面选择功能部分零件下没法指定面

649766048, 2024-11-20:
### Bambu Studio Version

1.10.1.15

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

CPU：Ryzen7 7840HS
显卡：RedeonT 780M

### Printer

A1/P1/X

### How to reproduce

导入后，使用选择底面功能，无法选择某一个平面

### Actual results

需要进行该底面选择，以避免大角度打印缺陷
[BUG.zip](https://github.com/user-attachments/files/17827885/BUG.zip)

### Expected results

将我选择的面作为底面，现在无法选择该面

### Project file & Debug log uploads

[BUG.zip](https://github.com/user-attachments/files/17827900/BUG.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-11-20:
We have noticed this issue and will upgrade this feature~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-11 by Haidiye00