bambulab/BambuStudio #5344: X-Y hole compensation in Bambu studio doesn't work correctly

mrootje, 2024-11-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Create a test piece with holes ranging from 1 to 10 mm in diameter. Print without hole compensation and print again twice, once with compensation for the largest hole, and one with compensation for the smaller holes.

### Actual results

When creating small and big holes, X-Y compensation either results in correctly sized small holes while larger holes become to large, or in large holes being correct while small holes stay to small. 
The correction should take into account the size of the holes.

### Expected results

G code generation should take into account the actual hole diameter and apply X-Y compensation dependent on it, in order to be able to produce items with correct diameters from small to big. Seems small holes need a relative big enlargement and big ones do not.

### Project file & Debug log uploads

[holetest.zip](https://github.com/user-attachments/files/17828128/holetest.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-26:
@mrootje '...Seems small holes need a relative big enlargement and big ones do not.' It may remind us this could be a new feature rather than a bug. Can you describe more about the XY shrink? Like is that the XY compensation should be a linear relationship with hole size? Or just a threshold will be better?

mrootje, 2024-11-26:
Not sure what you mean with new feature? What I'm trying to say is that a 1 mm hole requires a model with a 1.6mm hole. A 2mm hole requires a model with a 2,4mm hole, a 3mm hole requires 3.3mm, 4mm requires 4.2, 5 requires 5.2, 6-9 require 0.2 bigger, and 10 require 0.15 bigger... 
So when setting hole compensation, it is only right for a a small range of diameters and therefore not usable.