bambulab/BambuStudio #5346: select filaments 11-16 by number

UBaer21, 2024-11-20:
**Is your feature request related to a problem? Please describe.**
I can select filament 10 by pressing 1 then 0,  doesn't work for 11 ...

**Describe the solution you'd like**
select 11-16  same as i can select 10 by pressing the numbers

**Describe alternatives you've considered**
select by mouse dropdown

JohnSmithCYM, 2024-11-26:
[image]
Like the gif above, Press 1 and 2 in a short time get the purple cube,and 1 and 3 get green, is that what you want?

UBaer21, 2024-11-26:
Yes but it turns out this is not working always.
e.g if you have parts (not objects)

UBaer21, 2024-11-26:
These are objects not parts.
merge them into an assembly so you have parts.
[image]

JohnSmithCYM, 2024-11-26:
[image]

UBaer21, 2024-11-26:
interesting - i am using the objects browser to select the parts - then it is not working.
but i can confirm your way that it works when you click the parts. 
Further it doesn't seem to be an issue with objects and parts it is clicking on the plate view works but selecting via the object list doesn't.
[image]

JohnSmithCYM, 2024-11-26:
@UBaer21 I use the 'Alt+Left mouse' to select the indivisual part then press the number of filament.

UBaer21, 2024-11-26:
@JohnSmithCYM  Yes i understood that - but for complex multimaterial/color prints clicking the objects/parts is very inconvenient compared to going through a list of 10 parts and press the number of filaments (without mouse) - which works for filament slots 1-9 but not the other. Also selecting the items via mouse clicking in the list doesn't change that. 
I also have designs where 3D view clicking is impossible as the colored parts are embedded and not visible at all in the "prepare" tab (shine through) or at least require reorientation.