bambulab/BambuStudio #5349: External spool selection does not contain all Bambu Lab filaments

Woeters, 2024-11-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1 or X1

### How to reproduce

Go to 'Device', click on 'Ext. Spool'

### Actual results

It shows only a select set of filaments.
[image]

### Expected results

I expect the list to contain at least all Bambu Lab filaments, especially the filaments not suitable for AMS or AMS Lite:
- PLA-CF
- PLA Marble
- PLA Wood
- PLA Aero
- PETG-CF
- ABS-GF
- etc.

For future it might be easier to create a subsection (e.g. "PLA", "PETG", "ABS") to select from first, then second select which specific type of filament you're using (e.g. "Basic", "HF", "CF").

### Project file & Debug log uploads

There is no project or log file, simply start up Bambu Studio.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-11-21:
HI [Woeters](https://github.com/Woeters)：Please have a try following the steps [image]

Woeters, 2024-11-21:
That is strange, all specific filaments were checked in the “add/remove filaments” section so I just clicked ”Confirm”. After that, all filaments where available to add under “Devices” -> “Ext. Spool”It definitely needed a refresh for them to show up, the last time I added filaments in the “add/remove filaments” section must have at least 3-4 months ago. All filaments were checked back then.I cannot reproduce the issue now it’s updated, I will let you know if I can.Thanks for your help!