bambulab/BambuStudio #5354: Tree Auto ignoring "On build plate only"

nickm324, 2024-11-20:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1

### How to reproduce

1. Load a model that requires supports.
2. Enable supports.
3. Select On build plate only checkbox.
4. Slice
[image]

### Actual results

It adds supports to no build plate parts of the model.

### Expected results

Should ignore anything not touching the build plate.

### Project file & Debug log uploads

[TreeSupportIssue_-_camera.zip](https://github.com/user-attachments/files/17837572/TreeSupportIssue_-_camera.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tommyboy180, 2024-11-21:
Not a bug! This is the correct behavior!

Checking that box prevents supports from generating on anything other than the build plate. If you don't want supports to generate at the top, use the support blocker. 

1.10.1.50 [image]

1.9.7.52 
[image]

tangketan, 2024-11-27:
@tommyboy180 Thanks for your explanation.
I guess what @nickm324 means is the tiny bit touching the model when selecting "Tree Slim" style. That IS a quite old issue. We are working on it. Currently you can use "Tree Organic" style.
[image]