bambulab/BambuStudio #5356: Maximum layer height "too large layer height"

johnnydoesdesign, 2024-11-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1

### Additional system information

Apple MacStudio, M1 Ultra 20 core-CPU 48 core-GPU, 64GB RAM

### Printer

Bambu LAb P1S

### How to reproduce

1. Load model
2. in prepare > Quality set a layer height exceeding 0.6mm (eg. 0.64mm)
3. Get an error message: "too large layer height. reset to 0.2mm"
4. be annoyed ;) 

### Actual results

Theres a hard coded limit for layer heights at 0.6mm. 
This shouldnt be the case. Why would it be?

### Expected results

No hardcoded layer height limit.
Instead have an option in the printer settings, to limit layer height (max and min for that matter) to a resonable number depending on the installed nozzle size. 0.6mm doesnt make any sense as its too large for the standard 0.4mm nozzle but too little for a 0.8mm nozzle printing a 1.4mm wall thickness.

### Project file & Debug log uploads

[BambuStudio layer height issue.3mf.zip](https://github.com/user-attachments/files/17843368/BambuStudio.layer.height.issue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

clstr, 2024-11-21:
I loaded up your 3mf on the same version and it slices fine.

johnnydoesdesign, 2024-11-21:
have you tried changing the layer height to anything larger than 0.6mm though?

clstr, 2024-11-21:
> have you tried changing the layer height to anything larger than 0.6mm though?

Ah, apologies, I see the same error despite the extruder limits being set higher.

clstr, 2024-11-21:
This piece of code here seems relevent

https://github.com/bambulab/BambuStudio/blob/1540fbd9065d6b92246715ca40f4ec693c4807ea/src/libslic3r/Slicing.cpp#L42

Given a .8 nozzle and following the 3/4 guideline, .75*.8 = .6

johnnydoesdesign, 2024-11-21:
Mhh in. my view it shouldnt be dependant on the nozzle diameter but the line width as you can print significantly wider lines with a nozzle than it's bore is in diameter. 
Iam printing 1.4mm lines with bambulabs 0.8mm nozzle without issues, so for that linewidth a layer height of 1,05mm should be acceptable.

XunZhangBambu, 2024-11-28:
@johnnydoesdesign  That's a hard coded value. We should limit the adjustable layer height to between min_layer_height and max_layer_height on the machine params page.
[image]

johnnydoesdesign, 2024-11-28:
@XunZhangBambu 
That would be awesome! 
funnily the adaptive layer height feature already lets you paint in layer heights above the 0.6mm limit up to the limit set in machine > extruder settings.

pablogs9, 2024-12-14:
Same problem here. Using BS to slice for X1C but also for other custom machines with wider nozzle. Very annoying to have 1mm nozzle limited to .6mm layer height. 

Is this a planned feature? Would be great.