bambulab/BambuStudio #5358: Can't create custom filament since updating to version Bambu_Studio_win-v01.10.01.50

Ohman-BigO, 2024-11-21:
### Bambu Studio Version

01.10.01.50 

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU:13th Gen Intel Core i9-13900H 16GB

### Printer

Bambu Labs P1S

### How to reproduce

Select filament to use - Custom Filament - Create new 
No printers are shown and I can't select a Filament Preset

### Actual results

Error that no printer was selected, no filament is created

[image]

### Expected results

A custom filament should be added for use

### Project file & Debug log uploads

No Project file, because it doesn't matter what I try to print, I can't create a filament.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-22:
[image]
The gif shows the operation that I created a filament like your information with 1.10.1.50. It seems works well, so is there anything I missed?

ttt888zzz, 2024-11-22:
HI [Ohman-BigO](https://github.com/Ohman-BigO)：Please have a try following the steps.[image]

Ohman-BigO, 2024-11-22:
OMG I found the problem! Because I have no PLA-CF filaments created, it
doesn't have anything to show in the list. When I select PLA it finds some
already created and makes menu entries. In my opinion, this empty menu is
confusing and should at least show Bambu filaments. A menu that doesn't
open and doesn't bring an error appears broken.

Thanks for the help, close the ticket please.

Kind regards
Steve

Ohman-BigO, 2024-11-22:
Hi John, thanks for your answer, at least it showed me that the problem is
local.

I tried it again and still had an empty menu with PLA-CF selected. I chose
PLA and it showed me the other PLA filaments I had created, then it was
clear.

A blank menu is a bad thing, I would suggest that if no filament in this
class has been created, that it goes back to the standard filaments and
takes one from there.

Again, thank you for your assistance.

Kind regards
Steve

JohnSmithCYM, 2024-11-22:
You're welcome~

closed 2024-11-22 by JohnSmithCYM