bambulab/BambuStudio #5359: Sub-screen interface abnormality when dual-screen is displayed

lingyongji, 2024-11-21:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 23H2

### Additional system information

CPU: Intel i7 13700K
GPU: AMD 6700XT
Memory: 64G

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Open BambuStudio, the display of it is normal
2. Move Studio to the sub screen
3. The display of is messed

### Actual results

When move to sub screen, display messed.
[image]

### Expected results

When on the main screen, display normal.
[image]

### Project file & Debug log uploads

[project file & log.zip](https://github.com/user-attachments/files/17847135/project.file.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-22:
What are the resolutions of your main screen and sub screen?

JohnSmithCYM, 2024-11-26:
[image]
Confirm with your resolution.

lingyongji, 2024-11-26:
Main screen 4K and 150% scale
Sub screen 1080P and 100% scale
Looks like you reproduce the issue on you PC
[image]