bambulab/BambuStudio #5368: Import SVG - Contains Unsupported Error

MillerSC99, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

AMD Ryzen 9 7950X 16-Core Processor               4.50 GHz
32.0 GB (31.1 GB usable)

### Printer

A1 Mini

### How to reproduce

All the SVG files that I have imported in the past no longer work.  I just drop the SVG into the open app and it fails.  There is no way have it break the selection into objects.  

**The error**
[image]

**The file**
[image]

### Actual results

I get stuck and have to close it out, as the SVG imports as a single assembly and it is impossible to break it into objects.

### Expected results

I expect it to work as it used to.  I was able to make this model before using the same imported SVG.

**Link to model**
https://makerworld.com/en/models/807378?from=search#profileId-748075 

### Project file & Debug log uploads

[HoneyBadger Test.zip](https://github.com/user-attachments/files/17853992/HoneyBadger.Test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Haidiye00, 2024-11-22:
hello,due to some negligence in the feature upgrade, we will fix it as soon as possible.At present, you can import with the old version first, and then continue editing with the new version. Sorry for the inconvenience caused to you~

Haidiye00, 2025-01-10:
Hello, the new beta version has been released. Welcome to download and experience it~ 
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-10 by Haidiye00