bambulab/BambuStudio #5369: Automatische Flusskalibrierung dauert zu lange

FliegerRob, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Druck starten
2. Automatische Flusskalibrierung durchführen

### Actual results

Nach dem Start des Druckvorgang´s beginnt der Drucker mit der Automatischen Flusskalibrierung. Diese dauert teilweise über 4min und wird mehrmals ausgeführt. Dabei wird so viel Filament extrudiert, das die Nozzle manchmal komplett mit Filament bedeckt ist und es zu Fehlern führt.

### Expected results

Normale Kalibrierung

### Project file & Debug log uploads

alle Dateien betroffen

### Checklist of files to include

- [X] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-26:
Can you offer a log for us to analyze?

FliegerRob, 2024-11-26:
I hope these are the right files. I don't have that much experience.

 
 

Gesendet: Dienstag, 26. November 2024 um 08:43 Uhr