bambulab/BambuStudio #5371: Setup Wizard Fails when Choosing Filament After Update

SavvySpiro, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windowa 10

### Additional system information

Not a performance or rendering issue

### Printer

BambuLab A1 Mini

### How to reproduce

1. Open bambulab and setup wizard automatically opens
2. Select only A1 Mini 0.4 nozzle and move on
3. Leave all selected and press continue on the filament page

### Actual results

If I select or un-select any set of the options it will pop up with the same error message, and if I press either yes or no nothing happens. Restarting does not fix this issue. 

[image]

### Expected results

If items are selected, this error should not pop up. If nothing is selected, it should pop up but allow me to continue if I select yes to approve the defaults. 

### Project file & Debug log uploads

This happens for every file I open, including a brand new project. There is also no log file/crash file, because it just does nothing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SavvySpiro, 2024-11-22:
Forgot to include video of behavior. 
[Uploading video.webm…]()

lanewei120, 2024-11-25:
Hello @SavvySpiro 

Is this a 100% reproducible issue?
Does the same phenomenon occur when you select other models at the same time?

Please upload the work logs for reference. （guide: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)

Thank you!

realzorro, 2024-11-25:
Hello @SavvySpiro 
Could you please provide the logs of Studio,
This is the method for obtaining Studio logs: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Does this problem occur every time? Will the same problem occur on your other computers as well?

GitNitneroc, 2024-12-13:
Exact same issue on linux on version 1.10.1.50
This occurs every time and actually means I cannot print anything because I don't have a printer anymore in bambu studio

zhil, 2025-05-15:
the same issue with latest Bambu_Studio_ubuntu-24.04_PR-6688.AppImage - cant use Bambu Studio because of this bug.

scrawnymeasle, 2025-08-15:
Has a solution for this issue been found?

Pantanni, 2025-11-25:
I just unboxed and set up my new A1 and have this issue as the first experience with 3D printing... not fun

lsourdis, 2026-02-02:
Confirming exact same issue when installing fresh Bambu Lab Studio app on Windows 11 for my new P2S.