bambulab/BambuStudio #5373: SD Card is still on initialization

clemtt85, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to Device
2. Go to MicroSD Card
[image]

### Actual results

I cant see all of my saved codes files on sd card

### Expected results

I would reach all of my saved gcode files on sd card

### Project file & Debug log uploads

[Troubleshooting.zip](https://github.com/user-attachments/files/17867677/Troubleshooting.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

bambu123, 2024-11-26:
Check the network or restart the printer and try again.

clemtt85, 2024-11-26:
Restarting the printer worked !