bambulab/BambuStudio #5376: Piece on plate causing Bambu Studio to crash during slicing

AndersWestman, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 Enterprise, Version 22H2

### Additional system information

CPU: 12th Gen Intel(R) Core(TM) i7-12850HX
Memory: 32,0 GB
Display card: NVIDIA RTX A3000 12GB Laptop GPU
Bambu Studio zip-version (not installer)

### Printer

Bambu P1S

### How to reproduce

1. Open attached project file
2. Slice/prepare the plate
Note: There is a small piece on the plate that seems to cause the crash. It´s a missed leftover from an exploded stl, but it took some time to find as it was more objects on the plate at that time.

### Actual results

Bambu Studio crash during generating G-code

### Expected results

Bambu Studio should slice the plate or generate a message if the plate is not possible to slice correctly. 

### Project file & Debug log uploads

[Bambu crash issue.zip](https://github.com/user-attachments/files/17868419/Bambu.crash.issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

J0n-Sn0w, 2024-11-22:
Have the same issue. Sometimes small changes like reducing the amount of objects on the plate lets it slice it without issues. Its not very good reproduceable because adding the object back in the exact same position doesnt lead to the crash and the plate is also slicable. Very weird. I have this bug very often.

QingZhangBambu, 2024-11-26:
The empty first layer causes the crash. I will fix it.

QingZhangBambu, 2025-03-28:
[image] it had been fixed, please try the lateset version

AndersWestman, 2025-03-28:
Looks good. Thanks!