bambulab/BambuStudio #5377: Zooming out on latest update

mlcrabill, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows Enterprise 11

### Printer

Bambu Lab P1S

### How to reproduce

When I open Bambu Lab Studio and go to prepare - the build plate zooms way out. i can't move items and view it up close to be able to edit the items 

### Actual results

I try to zoom in and it automatically zooms out

### Expected results

I should be able to see the build plate up close

### Project file & Debug log uploads

na 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tommyboy180, 2024-11-23:
Works fine. This isn't a bug. 

https://github.com/user-attachments/assets/3572bb60-c29c-4e34-bfe0-8a1f4c6b8a51

mlcrabill, 2024-11-23:
Okay. Thank you for sharing. It’s bugging on mine (hence the request for support) and didn’t do that until I installed the updated. Michelle Crabill571-235-8342

https://github.com/user-attachments/assets/3572bb60-c29c-4e34-bfe0-8a1f4c6b8a51

mlcrabill, 2024-11-25:
> Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

It wasn’t working after the update so I uninstalled the software and reinstalled it and now it’s working fine. Michelle Crabill571-235-8342

Haidiye00, 2024-11-26:
Okay, thank you

closed 2024-11-26 by Haidiye00