bambulab/BambuStudio #5382: `Prepare` and `Prepare` are blank (not render) after upgrade latest version

wolf1111a, 2024-11-22:
### Bambu Studio Version

v01.10.00.89

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

MacOS 15.1

### Additional system information

CPU: 2.6 GHz 6-Core Intel Core i7
GPU: AMD Radeon Pro 5300M 4 GB / Intel UHD Graphics 630 1536 MB
Memory: 16 GB 2667 MHz DDR4

### Printer

X1C

### How to reproduce

1. download latest BambuStudio from official link.
2. install the dmg
3. open the BambuStudio
4. The `Prepare` and `Prepare` tabs are blank

Note: downgrade BambuStudio to old version not fixing the issue

### Actual results

[image]

### Expected results

expected to render the 3d model

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17873768/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-26:
Have you tried the newest version https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50 ? It may help and if you still failed ,you can try to close the "enable multi instance rendering by opengl"
[image]
[image]

guanyun-gudujian, 2024-12-26:
> ### Bambu Studio Version
> v01.10.00.89
> 
> ### Where is the application from?
> Bambu Lab Official website, Bambu Lab github releases
> 
> ### OS version
> MacOS 15.1
> 
> ### Additional system information
> CPU: 2.6 GHz 6-Core Intel Core i7 GPU: AMD Radeon Pro 5300M 4 GB / Intel UHD Graphics 630 1536 MB Memory: 16 GB 2667 MHz DDR4
> 
> ### Printer
> X1C
> 
> ### How to reproduce
> 1. download latest BambuStudio from official link.
> 2. install the dmg
> 3. open the BambuStudio
> 4. The `Prepare` and `Prepare` tabs are blank
> 
> Note: downgrade BambuStudio to old version not fixing the issue
> 
> ### Actual results
> <img alt="Screenshot 2024-11-22 at 09 12 39" width="1919" src="https://private-user-images.githubusercontent.com/189386126/389030232-3fd8eb04-68dd-4f9f-b680-3af00dc57843.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxODc5NTEsIm5iZiI6MTczNTE4NzY1MSwicGF0aCI6Ii8xODkzODYxMjYvMzg5MDMwMjMyLTNmZDhlYjA0LTY4ZGQtNGY5Zi1iNjgwLTNhZjAwZGM1Nzg0My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNlQwNDM0MTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZTI1OTM0YjYwZDIwOGQyMmYwYmI3NjBjYzFmMzdjODU5MGZjMmQwNWFhM2Q1NWY0MWM0ZDdkMGM5MmNlYmM2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.by4-dmc90ZIW7jYW9D575pe5HmKTZFKNZ2G2KJ5Ktm8">
> ### Expected results
> expected to render the 3d model
> 
> ### Project file & Debug log uploads
> [log.zip](https://github.com/user-attachments/files/17873768/log.zip)
> 
> ### Checklist of files to include
> * [x]  Log file
> * [ ]  Project file

hi, pls help to check if there is a context menu popped after right click in the 3d region, thanks.
like this:
[image]

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

Haidiye00, 2025-11-19:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.
If there are any further questions, let's continue the discussion. Thank you