bambulab/BambuStudio #5383: Media player (live stream and Timelapse's) not working in Studio

Teedabbeljoe, 2024-11-22:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.0.1

### Additional system information

M3 Pro version

### Printer

A1 Mini

### How to reproduce

Go to device status and open live camera feed. 
Go to device status and open SD card.

### Actual results

The live feed does not start, I am getting the error message: Player is malfunctioning. Please reinstall the system player.
In SD card viewer the error displays: Please check the network and try again. You can restart or update the printer if the issue persists. [-22005].

The live feed works on phone and iPad. The SD card shows the models in phone and iPad but not the Timelapse's.

### Expected results

Live feed should start the live feed stream from the printer.
SD card should show models and Timelapse's.

### Project file & Debug log uploads

Not relevant, happens with all files.

### Checklist of files to include

- [X] Log file
- [ ] Project file

miguesan01, 2024-11-24:
Same problem with MacBook Pro M4 and Sequoia.

bambu123, 2024-11-26:
Can you help me see if this is useful for you?

https://youtrack.jetbrains.com/articles/SUPPORT-A-564/macOS-15-Local-Network-Privacy-blocks-IDE-connections-to-local-resources-databases-app-servers-etc-unless-explicitly-permitted
[image: https://youtrack.jetbrains.com/api/files/334-61502?sign=MTczMTU0MjQwMDAwMHwxMS0xfDMzNC02MTUwMnw2OVNqc0F4VXM4OFMya1plQjUzT3IxZ1NITjFrTmRXX2FmSDRIdmpvdk9JDQo&updated=1727193593765]

AwakeProphet, 2024-11-26:
> Tudnátok segíteni, hogy ez hasznos-e az Ön számára? [https://youtrack.jetbrains.com/articles/SUPPORT-A-564/macOS-15-Local-Network-Privacy-blocks-IDE-connections-to-local-resources-databases-app-servers-etc-unless- kifejezetten engedélyezett](https://youtrack.jetbrains.com/articles/SUPPORT-A-564/macOS-15-Local-Network-Privacy-blocks-IDE-connections-to-local-resources-databases-app-servers-etc-unless-explicitly-permitted) [image: [https://youtrack.jetbrains.com/api/files/334-61502?sign=MTczMTU0MjQwMDAwMHwxMS0xfDMzNC02 MTUwMnw2OVNqc0F4VXM4OFMya1plQjUzT3IxZ1NITjFrTmRXX2FmSDRIdmpvdk9JDQo&updated=1727193593765](https://youtrack.jetbrains.com/api/files/334-61502?sign=MTczMTU0MjQwMDAwMHwxMS0xfDMzNC02MTUwMnw2OVNqc0F4VXM4OFMya1plQjUzT3IxZ1NITjFrTmRXX2FmSDRIdmpvdk9JDQo&updated=1727193593765) ]
> […](#)

Hi every one,

Same problem here.

Log file attached below.
[debug_network_Tue_Nov_26_09_35_03.log.zip](https://github.com/user-attachments/files/17916522/debug_network_Tue_Nov_26_09_35_03.log.zip)

closed 2024-11-26 by Teedabbeljoe