bambulab/BambuStudio #5384: Filament syncing with AMS does not work

gameengineer, 2024-11-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Ever since upgrading to the newest 1.10.x Bambu studio the syncing feature either does not work or is wrong. Used to work flawlessly.  Using the same filament with the same ID tags. 

### Printer

P1S

### How to reproduce

Turn on printer and Sync or Resync

### Actual results

Ever since upgrading to the newest 1.10.x Bambu studio the syncing feature either does not work or is wrong. Used to work flawlessly.  Using the same filament with the same ID tags.  Does not get the filament correct.

### Expected results

get the filament correct

### Project file & Debug log uploads

[log_2.zip](https://github.com/user-attachments/files/17878165/log_2.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tommyboy180, 2024-11-23:
I can't replicate. This isn't a bug. 

https://github.com/user-attachments/assets/98f36fb4-8e7e-4912-90f9-37ec00a51168

MartinChan3, 2024-11-25:
Have you opened the multi-device function? This could influence the sync result ...

bambu123, 2024-12-18:
From the log, there are 4 AMS materials. There are no abnormalities. Can you explain the problem points in more detail? It would be better to provide error screenshots.
ams 0 tray 0 id GFA05 color EAECEBFF
ams 0 tray 1 id GFL99 color F73CA0FF
ams 0 tray 2 id GFA00 color 000000FF
ams 0 tray 3 id GFA00 color FFFFFFFF