bambulab/BambuStudio #5386: Bambu Studio freezes when opening a STEP file. Only since version 1.10.1.50

Gregory-337, 2024-11-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sonoma 14.6.1

### Additional system information

Apple Silicon M1 
16gb RAM

### Printer

Bambu Lab A1

### How to reproduce

Import a STEP file to Bambu Studio. It will freeze.

### Actual results

Sometimes, after a longer time, it will work again. Sometimes it crushes, sometimes it freezes permenantly. 

### Expected results

It should open the file.

### Project file & Debug log uploads

Any STEP file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MackBambu, 2024-11-24:
How large is your STEP file ? Since reading STEP files occupies the main thread, there might be a freezing issue.

Gregory-337, 2024-11-24:
It also happens with very small files. I just tied one, that ist 63kb and it feezes. 

But I never had this issue with the previous version of Bambu Studio. So it must be a bug in the latest version.

MackBambu, 2024-11-24:
> It also happens with very small files. I just tied one, that ist 63kb and it feezes. But I never had this issue with the previous version of Bambu Studio. So it must be a bug in the latest version.

Have you tried different STEP files? Is this a common issue? If not, would you mind sharing this specific STEP file?

Gregory-337, 2024-11-24:
Yes I have tried various STEP files. I have the issue with all of them. 
Here ist one, for instance, that I created with onshape. No problem to open it with the last version of Bambu Studio. With the new one it won’t open. 

￼

MackBambu, 2024-11-24:
I can't see the file. You may need to upload it via the web interface.

Gregory-337, 2024-11-24:
I had to make Zip file out of it to upload. 
[Nangrip 1 schlitz unten.step.zip](https://github.com/user-attachments/files/17894959/Nangrip.1.schlitz.unten.step.zip)

JohnSmithCYM, 2024-11-26:
[image]
@Gregory-337 I have tried your step and you can see the gif above.

Gregory-337, 2024-11-26:
Thank you. Are you on a Mac or on windows? Von meinem iPhone gesendet

20241126151652_rec_.gif (view on web)

Gregory-337, 2024-11-26:
And I don’t get the „step file import parameters“ popup. Can I activate that? Von meinem iPhone gesendet

JohnSmithCYM, 2024-11-26:
@Gregory-337 On Windows11, and you can open the step dialog with this under Preference 
[image]

SergeyMokhov, 2024-12-07:
[Archive.zip](https://github.com/user-attachments/files/18050119/Archive.zip)
I have the same issue. Faucet does not load, Shovel Handle takes a second, but loads fine. 
15.1.1 (24B91) (M3), Bambu studio 10.10.0.89

MackBambu, 2025-04-29:
The new version has already switched to a different threading model. Does the issue still persist?

SergeyMokhov, 2025-04-29:
With the `Faucet.step` yes. Shovel takes a minute, but loads. 

[image]

SergeyMokhov, 2025-04-29:
Oh, sorry, I'm still on 1.10.1.50. Let me try an update

SergeyMokhov, 2025-04-29:
I'd say it still happens. `Faucet.stl` been spinning for over 10 minutes now on the latest release. Does not go unresponsive though, but still does not load. 

[image]

MackBambu, 2025-04-29:
> I'd say it still happens. `Faucet.stl` been spinning for over 10 minutes now on the latest release. Does not go unresponsive though, but still does not load.

Don't set the parameters too small, and try to keep the number of generated faces below 200,000; otherwise, it will take too much time.

https://github.com/user-attachments/assets/96cc1b10-738f-4356-a0d3-f8a6df5f13ac