bambulab/BambuStudio #5387: STEP model in the latest version are imported as a group of objects, preventing proper split for multi-color printing

wuyuanyi135, 2024-11-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Not relevant

### Printer

A1 MINI/ A1 Combo

### How to reproduce

Import the following STEP file: (zip due to upload limit)
[总装.zip](https://github.com/user-attachments/files/17879036/default.zip)

中文简述：旧版本STEP导入可以分割成零件，不会导致需要多色打印部分被拆散；新版本STEP导入直接被分成对象，无法分割成零件，导致分割时多色部件错位。

### Actual results

[image]
Imported STEP file are automatically disassembled into a group of objects, hence greying out the "split into parts" action. Splitting the objects results in misplaced objects. See the following screenshot for the letters falling down onto the platform.
[image]

### Expected results

In the previous version (1.9.7.52), the same model will be imported as a group of parts, hence splitting will not break them into the objects, making multi-color printing setting much simpler than the current version:
[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17879049/log.zip)
[project.zip](https://github.com/user-attachments/files/17879050/project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-11-24:
same #5214, merge issues

pdls3d, 2024-12-16:
Has this been resolved? I have run into the same problem with models that worked fine previously. Really looking for a repair.

cowboy, 2024-12-26:
Thanks for filing this issue, I'm experiencing the same problem with 1.10.1, and reverting to 1.9.7.52 fixed it.

pdls3d, 2025-01-11:
> Hi everyone, give the new version a try! A toggle option has been added
> when importing STEP files.
> https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64

Excellent. Thank you for update.

wuyuanyi135, 2025-01-11:
Thank you @MackBambu. Closing for resolution

closed 2025-01-11 by wuyuanyi135