bambulab/BambuStudio #5389: Long Retraction When Cut not working on A1 series

MeatKO, 2024-11-23:
### Bambu Studio Version

v01.10.01.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04.1 LTS

### Printer

BambuLab A1 with AMS

### How to reproduce

1. Download latest version of Bambu Studio
2. Pick an A1 or A1 mini printer profile
3. Place two primitives on the build plate, set distinct different colors for each of them e.g. : Bambu PLA Basic Black and White.
4. The print order should be by-layer so color swaps between Black and White can occur
5. Edit the Bambu PLA Basic profile  : 
Filament Settings -> Setting Overrides -> Enable "Long retraction when cut", enable "Retraction distance when cut" with default value of 18.
6. Either print or export the G-Code.

### Actual results

No "E-18", "E-1.8" or similar lines are observed in the source code.
The official "main" branch G-Code for filament change is dated as : "date: 20240830".
Upon visual inspection the extruder does not retract before cutting.
No difference in purge color gradients is observed. Proved experimentally by testing with and without the option enabled.

### Expected results

The expected result is to get the extruder to retract 18mm before cutting...

### Project file & Debug log uploads

[gopher_example_proj.3mf.zip](https://github.com/user-attachments/files/17879802/gopher_example_proj.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

The-Techy, 2024-12-24:
Same issue here on the A1 Mini. Left is printed with normal settings, middle with 'Long retraction when cut' enabled, and right with custom gcode from [https://makerworld.com/en/models/91241](https://makerworld.com/en/models/91241). Clearly, the 'Long retraction when cut' option does nothing, and I can see there is no retraction, especially compared to the custom gcode that adds it. You can see how much faster the colour changes.
[image]
[image]

Latest firmware and software (MacOS). Could this be a linux/MacOS issue, as it seems not many have this issue?

MeatKO, 2024-12-24:
@The-Techy  Most likely people just either don't know this option exists and/or don't test it once enabled.
the simplest fix was to add a 2-liner on the color change g-code right before the cut :
```gcode

;===== machine: A1 =========================
;===== date: 20240830 =======================
M1007 S0 ; turn off mass estimation
G392 S0
M620 S[next_extruder]A
M204 S9000
{if toolchange_count > 1}
G17
G2 Z{max_layer_z + 0.4} I0.86 J0.86 P1 F10000 ; spiral lift a little from second lift
{endif}
G1 Z{max_layer_z + 3.0} F1200

M400
M106 P1 S0
M106 P2 S0
{if old_filament_temp > 142 && next_extruder < 255}
M104 S[old_filament_temp]
{endif}

; below is my custom g-code
; change the length to your own liking, i've had 0 failures with regular, silk and glow in dark sunlu
G1 E-18 F200    ;Filament is pushed out 18 mm.
M400                  ;Waits until pushing out is completed before doing anything else. 
; end of custom g-code block

G1 X267 F18000
M620.1 E F[old_filament_e_feedrate] T{nozzle_temperature_range_high[previous_extruder]}
M620.10 A0 F[old_filament_e_feedrate]
```

The-Techy, 2024-12-24:
One would assume Bambu at least would test it, given they mention this feature in the [release notes](https://wiki.bambulab.com/en/a1/manual/a1-firmware-release-history) for the A1 Mini firmware updates

I'm just using the gcode from the aforementioned person on Makerworld, but looks like your solution would work too

XunZhangBambu, 2024-12-25:
@The-Techy Whether to apply long retraction is directly reflected in the flush matrix . After enabling the long retraction and recalculate the flush matrix, and you will notice that the flush volume decreases when the long retraction feature is enabled.

The-Techy, 2024-12-25:
@XunZhangBambu That doesn't seem right - when I tested it, I did not recalculate the flush matrix, but there was no retraction before cutting. If that setting is enabled, it should always do the retraction before cutting. If there is no retraction, there can be no decrease in flush volume.

Also, to test it, I just tried enabling long retraction, and recalculating the flush matrix. There was no difference in flush volume between before and after enabling it.

MeatKO, 2024-12-25:
@XunZhangBambu It did neither of these for me, nor did the extruder pull any filament, I spent 30 minutes observing it.

MeatKO, 2025-01-01:
An update to the A1 series printer profiles has been issued a few days ago
Just checked and it does retract the correct amount of plastic before cutting
Thank you @The-Techy @XunZhangBambu for the support, wish you a happy new 2025. Cheers.

The-Techy, 2025-01-02:
Just tested, it does indeed appear to be fixed! Thanks @XunZhangBambu, and thanks @MeatKO for pointing out the update!

XunZhangBambu, 2025-01-05:
@The-Techy The fixed configuration package has been released, so this issue can be closed.

closed 2025-01-05 by XunZhangBambu