bambulab/BambuStudio #5390: Spreadsheet brim does not match

kirk223, 2024-11-23:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.7.1

### Printer

BambuLab P1S

### How to reproduce

Brim type  does not work properly
Mark brim type but the actual brim type is wrong. For example:
Select outer brim - nothing happens when slicing
check the spreadsheet and the support info shows painted.
Select inner brim
the spreadsheet shows outer brim 
outer brim is produced when slicing
[image]

### Actual results

brim types are not matching properly and slices the wrong type

### Expected results

Selected brim should match up properly.  Happens on all files. The problem appeared when the painted brim was added.

### Project file & Debug log uploads

[Mrs. Claus_Torso_MC Kirk copy.3mf.zip](https://github.com/user-attachments/files/17880261/Mrs.Claus_Torso_MC.Kirk.copy.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-11-26:
confirm

kirk223, 2024-11-26:
Confirm

Kirk

nnobuddy, 2024-11-27:
Confirmed.

kirk223, 2024-11-27:
Im sorry but are you looking for additional information from me?

Kirk

nnobuddy, 2024-11-27:
> Im sorry but are you looking for additional information from me? Kirk
> […](#)

No, I just confirmed that your original report is true and that there is an issue in Bambu Studio.
Thx very much for reporting.

MackBambu, 2025-01-12:
fixed
try: https://github.com/bambulab/BambuStudio/releases/tag/v01.10.02.64

closed 2025-02-06 by MackBambu