bambulab/BambuStudio #5393: Erroneous supertack plate error

skopper1, 2024-11-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

Any print that contains PLA Silk that does not touch the build plate gives nonsensical errors about not using PLA Silk on the Supertack plate.  No kidding - but it's not an error when there is no PLA Silk on the first layer.

### Actual results

System forces you to set temps for PLA Silk above 0, but it's irrelevant.

### Expected results

This error should only appear when there is PLA Silk touching the build plate, i.e., on the first layer!

### Project file & Debug log uploads

[debug_Fri_Nov_22_16_12_34_21224.log.zip](https://github.com/user-attachments/files/17892054/debug_Fri_Nov_22_16_12_34_21224.log.zip)
[CribMan2 7 color board.zip](https://github.com/user-attachments/files/17892055/CribMan2.7.color.board.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-11-25:
@skopper1 Is this the warning message you get?
[image]

QingZhangBambu, 2024-11-25:
@skopper1 beside that,  i get no other warnings, could you provid a reproduce steps?
[image]

skopper1, 2024-11-25:
yes

 

image.png (view on web) <https://github.com/user-attachments/assets/29855e28-7b59-4007-902b-7c5d6e1acd7f>

skopper1, 2024-11-25:
> @skopper1 <https://github.com/skopper1>  Concerned staff said that Super Track does not support print pla silk. To print it, you must set the bed temperature above 0 and save the param.

You are not understanding.  There is NO pla silk touching the bed.  So there should be no warning.

QingZhangBambu, 2025-01-27:
@skopper1 I have conveyed your request to the relevant product manager, who will evaluate it.