bambulab/BambuStudio #5397: Mac Bambu Studio unresponsive

Nige7651, 2024-11-24:
### Bambu Studio Version

1.10.1.5

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 13.3

### Additional system information

Mac Mini M2 Pro

### Printer

Bambu Lab A1

### How to reproduce

App is very unresponsive, right click takes several clicks for the menu to appear. Add a build plate, takes many clicks to swap between them.. The 5 buttons alongside the build plate don't work.

### Actual results

The application is very unresponsive...

### Expected results

It should work! I have dual monitors, tried with 1 monitor and 2.

### Project file & Debug log uploads

[debug_Sun_Nov_24_10_35_29_1107.log.0.zip](https://github.com/user-attachments/files/17892921/debug_Sun_Nov_24_10_35_29_1107.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Nige7651, 2024-11-24:
I have found if I double click instead of single clicking, I can get some features to work...

MartinChan3, 2024-11-25:
I hope https://github.com/bambulab/BambuStudio/issues/5241 could help you.

Haidiye00, 2024-11-25:
Hello, we suggest you use a real mouse and test it again~

Nige7651, 2024-11-25:
I’m using a desktop Mac MINI M2 PRO - NOT a laptop…

Regards

Nigel Cartwright

07860 628106
***@***.***
https://www.modeltek.com

Haidiye00, 2024-11-25:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you

Nige7651, 2024-11-25:
[image]

I'm not sure where this gets you. I've greated a scond build plate and I'm clicking on it and the buttons at the side of it. THere's no response. Eventually they click will sometimes work.

I've disabled everything I can think of, I have a streamdeck which I've disabled and disconnected, still the same...  

Earlier versions of Studio have the same issue...

Haidiye00, 2024-11-25:
Hello, does a single disk get stuck?

Nige7651, 2024-11-25:
Stuck? no. I just have to keep clicking. The clicks are registering you can sometimes see the buttons change, but nothing happens. Eventually, usually a fast double click will register...

Haidiye00, 2024-11-25:
Hello, is it normal to click on the 5 buttons on the right side of the plate when there is only one plate

Nige7651, 2024-11-25:
Yes, but you're missing the ponit, I can't click between the plates - it takes 30 clicks to get the other plate to come into focus!

Haidiye00, 2024-11-25:
Hello, you can find the option for this preference setting and try turning it off to see if it works properly?
[image]
[image]

Nige7651, 2024-11-25:
Hi, yes I've tried that, no change... 

It's as though something is interceptng the clicks, the buttons seem to register the mouse is there and has clicked, but it takes many attempts before the reuest works. 

For instance if I add a second plate, then click on the X to delete it again, the 'X' box goes grey to register the cick but it might take twenty clicks for it to go... Here's another GIF, you can see the clicks registering in the 'X' boxes, but the plate dosen''t either come into focus or delete... 
[image]

Nige7651, 2024-11-25:
It's the same in version 1.9.7.52...

MartinChan3, 2024-11-25:
@Nige7651 So was there a version that you can use fluently in the past time? Maybe this will help Bambu to know more.

Nige7651, 2024-11-26:
Hi

I've been all the way back to 1.0.2, still the same problem...

Sorry... 

Here's a screenshot of what's running using Activity Monitor.
No idea if this helps...

I've tried running Studio after a restart with nothing else running.
Tried running uner single or dual monitors
Tried switching off Sttream Deck (Killed the process on Activity monitor too) 
Tried without the keyboard plugged in.

Still the same issue...

Nige7651, 2024-11-26:
[image]