bambulab/BambuStudio #5398: “Send print job to” dialog does not show filament-AMS mapping UI

capnslipp, 2024-11-24:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1 (24B91)

### Additional system information

MacBook Pro 16-inch, 2023, Chip: Apple M2 Pro, Memory: 32 GB

### Printer

Bambu Lab P1S + AMS

### How to reproduce

1. Download “P1S internally sliced files” from https://wiki.bambulab.com/en/general/bbl-printer-internal-sliced-files
2. Unzip “P1S-internal-sliced-files.zip” to “P1S_Gcode” folder.
3. Open “P1S_Scraper Grip by Bambu Lab_PLA_AMS.gcode” (or any of the gcode files) in Bambu Studio.
4. In the show Preview screen, click “Print plate” button in the top right.
5. Notice in the “Send print job to” dialog the error message “Please click each filament above to specify its mapping AMS slot before sending the print job”, yet not UI above exists to select filament-AMS mappings, and the Send button is grayed out.

Because of this, it is impossible to use Bambu Lab software to print a Bambu-supplied gcode file to the printer.

### Actual results

“Send print job to” dialog not showing filament-AMS buttons, with no way to proceed to printing:
[image]

Device screen, showing printer is recognized, and was able to successfully print Flow Dynamic calibration.

### Expected results

Example of properly-functioning “Send print job to” dialog, found on the Bambu forums.
[image]

### Project file & Debug log uploads

[Sun_Nov_24_13_56_00.log.zip](https://github.com/user-attachments/files/17894668/Sun_Nov_24_13_56_00.log.zip)
[P1S_Scraper Grip by Bambu Lab_PLA_AMS.gcode.zip](https://github.com/user-attachments/files/17894669/P1S_Scraper.Grip.by.Bambu.Lab_PLA_AMS.gcode.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MartinChan3, 2024-11-25:
It seems send 'Pure gcode' is quite different from send a 'Studio gcode', I believe https://makerworld.com/zh/models/45459?from=search#profileId-47205 is what you need.

walterwongbbl, 2024-12-03:
Hi @capnslipp 
We are currently unable to send individual G-code files to the printer; we only support 3MF format files.