bambulab/BambuStudio #5400: Display error in the print time summary

HolmHansen71, 2024-11-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 and A1 Mini

### How to reproduce

If you have several printing plates, simply click on “All Plates Stats” in the preview

### Actual results

[image]

### Expected results

The text for the individual printing plates should be slightly further apart.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17901719/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

JohnSmithCYM, 2024-11-25:
[image]
Can you offer a 3mf for me to test?And which screen resolution and scale ratio does you use in this case?

HolmHansen71, 2024-11-25:
[dividing_head_no_ams_with_bug.zip](https://github.com/user-attachments/files/17904843/dividing_head_no_ams_with_bug.zip)
Screen resolution:
[image]
I have reduced the project somewhat, as not everything is ready for publication yet. But the bug is already in here.

MartinChan3, 2024-11-26:
[image]
Maybe a little wider box will help to solve this problem