bambulab/BambuStudio #5402: Add buttons (OK/Save/cancel) confirming/discarding actions

2023gh, 2024-11-25:
This is more QoL feature. Some of your configuration pages don't have any buttons at all - just icon to close the window. 
Examples: 
Preference page:
[image]

Multi device config page:
[image]

Can you add OK/SAVE    + cancel buttons?

MartinChan3, 2024-11-26:
In my view,  Studio's save operation is acceptable because modern App (especially in phone) has used the same way for a long time. Maybe a vote for this problem should be held.

2023gh, 2024-11-26:
> In my view, Studio's save operation is acceptable because modern App (especially in phone) has used the same way for a long time. Maybe a vote for this problem should be held.

Save and cancel allow you to apply/revert options you've selected in the current session. You have to scroll page with options cause there many of them.  Assume you have only 1 option selected. Opened option page selected all remaining options and then changed your mind - you have to visit all options again and uncheck them while can just press cancel button.

Again, that is quality of life and not something urgent.