bambulab/BambuStudio #5403: mouse clicks ignored

Hofmangearsupply, 2024-11-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.1.1 

### Additional system information

new Mac mini m4 . 
fresh install with new computer - switching from windows 11.
Apple Magic Mouse.
Not Project specific.
Mouse clicks are ignored when selecting model on build plate or switching to another build plate. Sometimes it takes 3, 4 clicks (slowly) before input is registered. Same issue with selecting build Plate title or menu buttons along right side of build plate. Top Menu and Left side menu appears to be unaffected.

### Printer

All printers - I have them all.

### How to reproduce

1. open project
2. select model on build plate 
3. add build plate
4. select second build plate
5. switch back to first active build plate

### Actual results

2. - does not register immediately the first time or right click model to bring up submenu *eg. Clone object
4.  - does not register immediately. 
5.  - does not register immediately.

### Expected results

mouse clicks should register immediately. - not repeated slow clicks before selection is registered by software

### Project file & Debug log uploads

does not apply to issue. None project specific.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-11-26:
[image]
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
Is it normal to follow my operation?

Hofmangearsupply, 2024-11-26:
Is this a MacOS app?

Hofmangearsupply, 2024-11-26:
I will upload a recording using Screenshot native app on MacOS, but without the recorded mouse clicks, I'm not sure the video will be convincing to watch.

https://github.com/user-attachments/assets/813597d7-6495-4782-8b47-bd76bfcdf245

Hofmangearsupply, 2024-11-26:
here is a recording using Licecap - a Mac screen to gif app...
[image]
The system seems to be Lagging , sometimes it will register the mouse click right away. other times it takes 3 to 4 attempts . Not consistent in any way

Haidiye00, 2024-11-26:
open project
select model on build plate
add build plate
select second build plate
switch back to first active build plate

Hello, could you please record a GIF for the step you mentioned?

Andtry pressing Ctrl+P shortcut key, find this option to close, and try again~
[image]
[image]

Hofmangearsupply, 2024-11-26:
in my settings menu  (cmd + ,) the "enable multi instance rendering by OpenGL" is active, must be active by default because I have not changed those settings when I installed bambustudio on this new computer.
[image]

the issue is clearly only affecting the build plate area. All other menu items are responsive with mouse clicks. There is another user on GitHub that has commented on this which caused me to submit a support ticket. thank you

Haidiye00, 2024-11-27:
Hello, what did you do? Currently, it seems that your disk switch is normal

Hofmangearsupply, 2024-11-27:
Hello, my disk switch? I"m sorry I don't know what you mean. I will try and show the issue again. It is only affecting the mouse clicks when selecting build plates or models on those build plates. the other menu selections are normal

Hofmangearsupply, 2024-11-27:
[image]
you can clearly see that when trying to select the build plate it takes at least two mouse clicks before the input is registered.

Hofmangearsupply, 2024-11-27:
https://github.com/bambulab/BambuStudio/issues/1481

Hofmangearsupply, 2024-11-28:
Hello again. I just purchased a trackball mouse - Nulea M501 to try out. From what I can tell in using it for the last 10 min. is that the issue is with Apple Magic Mouse and the touch sensitivity with the surface of the mouse - it seems that the software is registering multiple inputs at once - because you can zoom with your finger like a track pad. This is something that should be addressed in the code as it doesn't behave this way with other programs. But this is at least a starting point. Maybe the software team can look into the compatibility of BambuStudio MacOS 1.10.1.50 and the Apple Magic Mouse.

Haidiye00, 2024-11-28:
Hello, thank you for providing the information. We will analyze and consider it~

plymouthvan, 2024-12-03:
Just wanted to chime in and say that I have this same issue with the Magic Mouse. It's been a problem since I got the first printer more than a year ago and it's insanely frustrating.
[image]
It definitely seems related by some kind of input overload where it thinks I'm trying to scroll or pan and is ignoring the clicks.

Hofmangearsupply, 2024-12-03:
Thanks for the comment, I definitely agree with you. I get the feeling that MacOS isn’t a priority to the software team at Bambu.. I have found it to be a lot more “buggy” compared to the windows version which I’ve been using since the Kickstarter launch. Windows Bambustudio is solid. In the meantime, I’ve returned my Magic Mouse and am using a nebula m501 trackball mouse and it’s been good as long as I don’t touch the ball when clicking….

guanyun-gudujian, 2024-12-30:
@Hofmangearsupply 
Hi, pls enable "Tap to click" under the system setting menu. like follow image:
[image]

than, tap (attention pls, not click here) on the trackpad to pick the print-bed and check if it is work or not.

thanks

plymouthvan, 2024-12-30:
@guanyun-gudujian I am using a Magic Mouse which doesn't have this setting in system preferences. However, I was able to configure a tap-to-click setting with BetterTouchTool, which does resolve the problem in BambuStudio at the cost of not being able to click, which makes it an untenable solution.

Also, to be sure, I disabled BetterTouchTool entirely to be certain it wasn't causing conflicts and the Bambu Studio click behavior was still broken.

guanyun-gudujian, 2024-12-31:
@plymouthvan 
ok, got it, thanks

plymouthvan, 2025-08-30:
This issue remains unresolved.

Hofmangearsupply, 2025-08-30:
Agreed. Unresolved. Apple Magic Mouse and Track Pad do not work properly.

plymouthvan, 2025-10-22:
It's mind boggling that this issue is still unsolved. Please put some resources into fixing this, especially since firmware updates are making it less convenient to use third party software.