bambulab/BambuStudio #5405: Serializing NaN error when slicing a plate

ogmueller, 2024-11-25:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.1.1

### Additional system information

MacBook Pro 16" 2021 32GB RAM

### Printer

Bambu Lab A1

### How to reproduce

I used a .3mf file and changed "First layer filament sequence" on "Plate Settings". Then hit "Slice Plate" and an error occurred. Changing back the sequence will solve the error, but not solve the problem.

### Actual results

When slicing this plate, i get a "Serializing NaN" error.

### Expected results

It should be sliced and filemant sequence should be taken into consideration.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17908986/log.zip)
[BugReport-NaN.3mf.zip](https://github.com/user-attachments/files/17908989/BugReport-NaN.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

jburridge88, 2024-12-19:
I have discovered a workaround for this error that may also give insight into the root cause of the problem - when setting the customised order for *other layers*, the layer range is autofilled to be '2' to '2147483647' (max 32-bit signed integer). This gave a "Serializing NaN" error when trying to slice. Reducing the upper bound to a more reasonable number (I believe the largest number of layers possible with a 0.2mm nozzle on a 256mm height range is somewhere around 6,400) eliminated this error.
 
I noticed in your example 3mf that you also set the other layer order, not just the first layer - and this 2147483647 upper bound was present. When I dropped that value to 10000, the custom order sliced perfectly fine. Presumably somewhere in the code the value is passed through something smaller than 32 bit and it fails.

TL;DR = It's an integer overflow issue, drop the max limit to a sane value and the problem is fixed.

thereseLYR, 2025-01-07:
To add on to this, the default upper bound (YY) of the 'Layer XX to YY' option seems to be initialized to `End` sometimes - I encountered this while trying to replicate the issue. If there is such a keyword for this, why not set the default option, or at least set the default option to a smaller or more reasonable number of 10000 like @jburridge88 suggested?
[image]

On the other hand, I'm not sure why the 'Layer XX to YY' option even exists. If I select the "Other Layers filament sequence" option instead of "First Layer filament sequence", my intuitive understanding is that I would be configuring ALL other layers (i.e. layer 2 to layer 99, 999), instead of having to set a range. 

In my opinion, the range is not useful because it does not allow multiple selection ranges e.g.
- Layer 2-50: Black > White
- Layer 51-100 (End): White > Black
Instead, the current implementation only allows for one range to be modified, i.e. I can apply `Layer 2-50: Black > White` OR `Layer 51-100: White > Black`, but not both at the same time. I believe that in most scenarios, a user would want to control the filament sequence for all other layers, and not just a range.

zhimin-zeng-bambulab, 2025-04-07:
@ogmueller  After you open the Plate settings dialog, click OK and it will be updated with the correct data.
[image]