bambulab/BambuStudio #5406: 自建耗材基本信息丢失，

Griselda-droid, 2024-11-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

WIN11

### Printer

P1SC

### How to reproduce

自建耗材使用一段时间后会丢失设定打印机
[image]

### Actual results

导致AMS中无法选择，以及编辑K值。其中最诡异的是在手机APP上依然可以选择该耗材，只是不能编辑K值

### Expected results

### ### 
进入配置文件，手动给耗材添加打印机后恢复正常
[image]
[image]

### Project file & Debug log uploads

[Filament presets.zip](https://github.com/user-attachments/files/17911981/Filament.presets.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-11-26:
在手动新增了compatible_printers之后，你这边能稳定复现这个问题吗？

Griselda-droid, 2024-11-26:
> 在手动新增了compatible_printers之后，你这边能稳定复现这个问题吗？

找到了稳定复现的操作，先打开一个3MF文件修改AMS里的耗材信息。保存然后打开另外一个3MF点击复原小按键就会清理掉打印机配置

[image]

MartinChan3, 2024-11-29:
我用你的这个预设文件试了好几次都复现不了，兄弟能搞个gif或者录屏看看操作流程吗

DanBao-Bambu, 2024-12-09:
I cannot reproduce it either, I don't know if the steps are different from yours.

Griselda-droid, 2025-04-02:
> 我用你的这个预设文件试了好几次都复现不了，兄弟能搞个gif或者录屏看看操作流程吗

Griselda-droid, 2025-04-03:
> I cannot reproduce it either, I don't know if the steps are different from yours.

I uploaded the video on Google, I hope you can take a look. Thank you.
https://drive.google.com/file/d/1ID7HnQ-v8RjCm-jimIsM64o2tOYr0ayw/view?usp=sharing

xiayanfire, 2025-07-10:
我也遇到了，bambu Studio 2.1.1.52，手动保存配置文件也没用。这个bug，太恶心了。