bambulab/BambuStudio #5407: Curve Text

shanmanf4, 2024-11-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Processor	Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz, 2112 Mhz, 4 Core(s), 8 Logical Processor(s)

### Printer

Bambu P1S

### How to reproduce

GO to T for text , type the text I wish to put on object , ty to click surface and Horizonal text 

### Actual results

Can not click Surface and Horizonal text at the same time. Text will not follow the shape of the object selected.  

### Expected results

I have been able to in previous versions of bambu studio to select Surface and Horizontal text to make the text follow the coverture of the object. 

### Project file & Debug log uploads

[Bowling Ball Greg JohnsCup(3).zip](https://github.com/user-attachments/files/17912439/Bowling.Ball.Greg.JohnsCup.3.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

JohnSmithCYM, 2024-11-26:
See https://github.com/bambulab/BambuStudio/issues/5308.

Haidiye00, 2024-11-26:
Hello, we have fixed this issue and it is currently undergoing internal testing. Please wait for the next version~

Haidiye00, 2025-01-11:
Hello, the new beta version has been released. Welcome to download and experience it~
https://github.com/bambulab/BambuStudio/releases

closed 2025-01-11 by Haidiye00