bambulab/BambuStudio #5411: Inconsistent results when Slicing

NVNDO, 2024-11-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambulabs P1S

### How to reproduce

Beforehand: 
Both Parts are similar in size and have the same height. 
Both have been created with Fusion 360.
Both have 0.8mm wide Bodies located at their bottom.

Import files into different BambuStudio windows (put them side by side) - Slice and lower the layer-bar until reaching the 1st layer. 

### Actual results

- The Square will show its bottom line, ring and dot
- The Hexagon does not.

Why?!

### Expected results

- If Square show its bottom line, ring and dot
- The Hexagon should do the same

PD: Usually I make a cutout if a line, ring, dot or whatever should be present in the first layer, BUT the design requires not doing so, therefore I tried if this would work, and it does - but ONLY sometimes. So something is off here. It should either never show the bottom lines, or always, but not be random. 

### Project file & Debug log uploads

[First line not showing.zip](https://github.com/user-attachments/files/17925263/First.line.not.showing.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Nikonguy60, 2024-11-27:
When I upload a picture from hue forge to print and I have selected my colours, after slicing the only colour it shows it black and not the rest

Nikonguy60, 2024-11-27:
That was not helpful

NVNDO, 2024-11-27:
It's because you did not change the color in objects (see screenshots). When posting here I assume people know how to use the slicer. I don't wanna be rude, but if you can not help, pls don't comment. Cheers

Nikonguy60, 2024-11-27:
Well just for the record I do know how to use the soccer and I did change
the colours. I think you were rude and I don’t appreciate that. There are
nicer ways of messaging people. Thank you very much!!

zhimin-zeng-bambulab, 2024-11-28:
When multiple parts are coplanar on the same plane, we will give priority to the color of the part listed later in the object list. For the example in the picture, you can drag the black part below the orange part in the object list.
[image]

Nikonguy60, 2024-11-28:
Ok, thanks.

NVNDO, 2024-11-28:
@zhimin-zeng-bambulab Oh wow... That is an unexpected & easy solution - I would probably have never tried that. Thank you so much!! 👍😊

closed 2024-11-29 by zhimin-zeng-bambulab