bambulab/BambuStudio #5412: Transition between speed of layers

Bush-Man72, 2024-11-26:
Currently if there is a speed discrepancy between layers there is an obvious line where the speed changes due to the different flow rates.
A solution could be to add a setting similar to the speed discontinuity area but instead have the speed of the previous or future layer in mind.

The only current alternative is to slow the whole external walls down which takes more time.

QingZhangBambu, 2024-11-29:
@Bush-Man72 I've tested that, and I found that the unsmooth surface is not caused by the different speeds of the outer wall but by the filament state while changing speed from infill to outwall.

QingZhangBambu, 2024-11-29:
@Bush-Man72 smooth speed between layers works on some cases, but notthe  root solution

QingZhangBambu, 2025-04-03:
@Bush-Man72
I hope you have fun with that.  It's available on the latest version

[image]