bambulab/BambuStudio #5413: Filament status not displaying correctly on AMS in Bambu Studio device window

Penguin-fach, 2024-11-26:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro: 10.0.19045

### Additional system information

Processor	Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz, 3901 Mhz, 4 Core(s), 8 Logical Processor(s)
Installed Physical Memory (RAM)	16.0 GB
Total Physical Memory	15.0 GB
Available Physical Memory	2.39 GB
Total Virtual Memory	22.3 GB
Available Virtual Memory	2.62 GB
Page File Space	7.33 GB
Name	NVIDIA GeForce GTX 660

### Printer

Bambu Lab X1C

### How to reproduce

1. Loaded filament into the AMS (3 x Bambu; 1 x generic)
2. Verified that the filament loaded correctly and registered on the printer. Entered details for the generic filament - all OK
3. Started a new project.
4. Loaded a STEP part, selected the filament (Bambu PETG-HF, grey) and sliced - all OK
5. Sent the job to the printer.
6. Opened the device. At the beginning of the print, when the filament is loading, the AMS status highlights the external filament (not presently used) and the coloured line of filament does not display correctly. Part of the coloured band is missing. Note also that the filament DOES load correctly. Just the display is incorrect.
7. If the page is minimised and reopened then the coloured band showing the filament loaded is now correct

### Actual results

[image]
[image]

### Expected results

The display was fine prior to the last update.
Bambu Studio does not crash so no log file so far as I am aware.

### Project file & Debug log uploads

[Simulated Lamp - Cable Grip - Rev D.zip](https://github.com/user-attachments/files/17926005/Simulated.Lamp.-.Cable.Grip.-.Rev.D.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

melted98, 2024-12-09:
Same Problem/bug here

DavidO911, 2024-12-12:
Exactly the same problem here. Since last update. Select another tab and then return to Device tab and the graphic corrects itself. I know of 3 other people with the same issue.

walterwongbbl, 2025-01-01:
@Penguin-fach  @melted98  @DavidO911 
We apologize for the inconvenience; this seems to be caused by insufficient UI refresh rate, and we will fix this issue.