bambulab/BambuStudio #5416: Nach Firmwareupdate immer Seriennummer eingeben

Growwoolf, 2024-11-27:
### Bambu Studio Version

1.10.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Printer

BambuLab X1C

### How to reproduce

Nach jedem Öffnen/starten von Bambu Studio und senden an den Drucker muss die Seriennummer eingegeben werden. Ansonsten kann man keine Dateien zum Drucker übertragen. 

### Actual results

[Dok1.pdf](https://github.com/user-attachments/files/17931062/Dok1.pdf)
Ohne Eingabe der Seriennummer kann man nichts mehr an den Drucker senden. 
Log Netzwerktest:
Wed Nov 27 09:11:53:test bambulab start...
Wed Nov 27 09:11:53:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=01.10.01.50&guide_version=00.00.00.01
Wed Nov 27 09:11:53:test bing start...
Wed Nov 27 09:11:53:[test_bing]: url=http://www.bing.com/
Wed Nov 27 09:11:53:test storage start...
Wed Nov 27 09:11:53:[test_oss]: url=http://upload-file.bambulab.com
Wed Nov 27 09:11:53:test http start...
Wed Nov 27 09:11:53:test storage upgrade start...
Wed Nov 27 09:11:53:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Wed Nov 27 09:11:53:test plugin download start...
Wed Nov 27 09:11:53:test storage download start...
Wed Nov 27 09:11:53:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=01.10.01.00
Wed Nov 27 09:11:53:test bing ip resolved = 2.16.204.144
Wed Nov 27 09:11:53:test bing ok
Wed Nov 27 09:11:53:test http ok
Wed Nov 27 09:11:53:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/01.10.01.01/8f52eddd78/win_01.10.01.01.zip
Wed Nov 27 09:11:53:test bambulab ip resolved = 172.64.152.215
Wed Nov 27 09:11:53:test bambulab ok
Wed Nov 27 09:11:53:test storage failed
Wed Nov 27 09:11:53:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Wed Nov 27 09:11:53:test storage upgrade failed
Wed Nov 27 09:11:53:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Wed Nov 27 09:11:53:downloading 5%
Wed Nov 27 09:11:53:downloading 10%
Wed Nov 27 09:11:54:downloading 15%
Wed Nov 27 09:11:54:downloading 10%
Wed Nov 27 09:11:54:downloading 20%
Wed Nov 27 09:11:54:downloading 25%
Wed Nov 27 09:11:54:downloading 30%
Wed Nov 27 09:11:54:downloading 20%
Wed Nov 27 09:11:54:downloading 35%
Wed Nov 27 09:11:54:downloading 40%
Wed Nov 27 09:11:54:downloading 30%
Wed Nov 27 09:11:54:downloading 45%
Wed Nov 27 09:11:55:downloading 50%
Wed Nov 27 09:11:55:downloading 55%
Wed Nov 27 09:11:55:downloading 40%
Wed Nov 27 09:11:55:downloading 60%
Wed Nov 27 09:11:55:downloading 65%
Wed Nov 27 09:11:55:downloading 50%
Wed Nov 27 09:11:55:downloading 70%
Wed Nov 27 09:11:55:downloading 75%
Wed Nov 27 09:11:56:downloading 80%
Wed Nov 27 09:11:56:downloading 60%
Wed Nov 27 09:11:56:downloading 85%
Wed Nov 27 09:11:56:downloading 90%
Wed Nov 27 09:11:56:downloading 70%
Wed Nov 27 09:11:56:downloading 95%
Wed Nov 27 09:11:56:downloading 100%
Wed Nov 27 09:11:56:test plugin download ok
Wed Nov 27 09:11:56:downloading 80%
Wed Nov 27 09:11:56:downloading 90%
Wed Nov 27 09:11:57:downloading 100%
Wed Nov 27 09:11:57:test storage download ok
Wed Nov 27 09:12:13:test storage start...
Wed Nov 27 09:12:13:[test_oss]: url=http://upload-file.bambulab.com
Wed Nov 27 09:12:13:test storage failed
Wed Nov 27 09:12:13:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Wed Nov 27 09:12:21:test storage upgrade start...
Wed Nov 27 09:12:21:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Wed Nov 27 09:12:21:test storage upgrade failed
Wed Nov 27 09:12:21:status=404, body={"error_msg":"404 Route Not Found"}
, error=

### Expected results

Behebung der kontinuierlichen Eingabe der Seriennummer.
Da die Daten ja in der Bambu Config enthalten sind, Seriennummer und Code.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/17931208/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Growwoolf, 2024-12-04:
Version 01.08.02.00

MartinChan3, 2024-12-04:
It's wired, I found my X1C is the same firmware (it is the newest public release) and now it doesn't show anything wrong, and can you reproduce the problem these days?It's annoying I know because too many people are using 3rd-party filaments.